FaradayRF/FaradayRF-Hardware #6: Synchronous/Asynchronous CC430 Radio Support

kb1lqd, 2016-10-23:
# Overview
It is likely that in the future Faraday will split into two main projects:
- A smart node digital radio with radio and communication processes controls on the CC430
- A (mostly) dumb radio module that allows a more powerful processor perform the communications such as a Raspberry Pi. Interactions with Faraday in this mode would largely be that of a raw modem interface through UART/SPI/CC1101 GDx pins

Texas Instruments provides a basic overview of the radio operations possible with the CC430: [http://www.ti.com/lit/an/slaa465c/slaa465c.pdf](http://www.ti.com/lit/an/slaa465c/slaa465c.pdf)

kb1lqc, 2016-10-24:
Upon inspection, I'd much rather use signals already traveling to the general CC1190 and GPIO pin area so I minimally affect my routing scheme. This means initially I'd like to perform the following IO changes:

- P3.0 to P4.5
- P3.1 to P4.6
- P3.2 to P4.7

This forces the CC1190 control signals to be moved over to GPIO channels. Since these signals generally route to the same area of the PCB this then results in minimal impact to the design.

kb1lqd, 2016-10-24:
@kb1lqc OK that makes sense. I should double check that all P4 GPIO do infact work as intended.

kb1lqc, 2016-10-24:
OK I updated and swapped the mentioned P3 and P4 signals. This was much easier to route since the channels all went to the same area of the PCB. This was changed in 4786a6094c6b3ada832c1a492106f61978195773.

kb1lqc, 2016-10-27:
Closing this issue as it was addressed in Issue #1

closed 2016-10-27 by kb1lqc