FaradayRF/FaradayRF-Hardware #27: Test GPS without inductors using increase data throughput

kb1lqd, 2016-11-13:
Re-test #14 with code that sends lots of RF data to verify.

kb1lqd, 2016-11-16:
@kb1lqc I assume to test this I can just remove and short over all the removed inductors like you did?

I plan on testing several more in-depth configurations:
- Basic firmware update to force constant RF TX buffer data to push unit as-is as fast as it can go indefinitely. I will first look for the GPS lock to remain after several minutes. I can also verify GPS data, not sure if I'll get telemetry during TX while jerry-rigging this.
- Initiate a constant output IDLE or carrier and check the GPS lock LED / data remains as well

kb1lqc, 2016-11-17:
@kb1lqd Yes you could just short over them. Probably should remove the caps though too.If you want I can give you my Faraday unit with these modifications Thursday. If you create this mode then it's simple to just wait a day and test it Thursday evening.

kb1lqd, 2016-11-17:
# Continuous TX Program

I made a really basic continuous transmit program using the standard Faraday RF stack after messing around with the continuous TX methods of the CC430 hardware settings and failing. This basic branch directly fills the RF layer 2 FIFO every housekeeping cycle with a new packet (blank/random data) and thus due to it's speed the FIFO should be full nearly always. This may not be fully continuous but is very very close. The button on the PCBA toggles a bit that enters this mode and allows continuous transmission THIS IS A TEST MODE AND SHOULD NEVER BE PULLED INTO MASTER!

I set the unit to 914.5MHz and an RF power setting of 152 which corresponds to about 500mW. Letting this unit transmit continuously should allow us to see if any power quality issues are observered or if the GPS with and without RF LC filters is affected by the transmissions.

kb1lqd, 2016-11-18:
# Test-2 - SN1 In REV D1 Configuration

I ran the transmissions for a few minutes at full power with the GPS locked and running proxy to debug observe the effects of REV D1 configuration and full power GSM at 914.5MHz. I attached a ~6 inch 9V battery harness and battery and reran the test and the GPS lock light remained flashing indicating good lock again.

I reran the test at 927.5MHz and passed with both USB and battery power.

I reran the test at 902.5MHz and passed with both USB and battery power.

I'm considering this testing issue resolved!

closed 2016-11-18 by kb1lqd