FaradayRF/FaradayRF-Hardware #49: Test External Power (9V) Cable Connected to MOSFET Connector P3

kb1lqc, 2017-01-22:
Title says it all. Analysis indicates that Ground would be  connected to the MOSFET drain and VCC would be connected to VCC correctly. This was done by design. At most I expect the body diode of the MOSFET to allow conduction and powering of Faraday. Passing 30mA during GPS + RX should be fine while transmitting could cause damage to the MOSFET. The groundshift could cause unexpected operations but should not cause damage.

kb1lqc, 2017-02-15:
OK I'm revisiting this. I've attached a Faraday Rev B (Purple PCB) SN 3 to an external 9V supply.  The VCC input is very similar. SN 3 does not have Rev B U11 populated while SN 1 does. I would like to keep this as similar as possible.

**Rev B**
[image]

**Rev D1**
[image]

And the connectors are the same for VCC and MOSFETs

**Rev B**
[image]

**Rev D1**
[image]

kb1lqc, 2017-02-15:
The MOSFET circuit is the same as well.

**Rev B**
[image]

**Rev D1**
[image]

kb1lqc, 2017-02-15:
By design I suspect that the MIC94072YC6 on Rev B will not stop any reverse conduction. However it is coming from USB so we shouldn't need to worry as USB is very polarized, you would need to really try to reverse that polarization.

The VCC input is protected the LTC4412 with the auxiliary PMOS (Q4 on Rev B and Q3 on Rev D1). The LTC4412 should protect against a reverse polarity by turning off the PMOS and the internal body diode will be reverse biased.

The immediate question is whether connection of VCC power to the MOSFET will cause damage. By design the polarized connections are the same but force VCC to always be Pin 1 on the power connection as well as MOSFET. No issue here. The GND will however attach to the NMOS drain. The body diode will be forward biased as GND will be attached to the cathode as shown above. This means that the effect will be the same as placing a diode between ground and a battery.

**Pros**

- Faraday should actually still work
- No damage should result unless we violate the NMOS transistor body diode power rating.

**Cons**

- Faraday's ground will be a diode drop above battery ground (DGND and shield)
- Could cause ground loops and extra current drain when connected to externally grounded items.
- NMOS will dissipate heat during normal operation of Faraday when it shouldn't be.

Actually, I am not totally sure what will happen if the NMOS is turned on while VCC is plugged into the MOSFET connection. At first look, the worst-case is the NMOS is non-function because gate voltage may not get high enough and best-case the diode drop goes toward 0V due to Rdson with a saturated NMOS!

IRLML6346TRPbF Transistor
[Datasheet](http://www.infineon.com/dgdl/irlml6346pbf.pdf?fileId=5546d462533600a401535668a336262e)

This can withstand 1.3A continuous through the body diode... this means Faraday without modifications or external current sinking should cause no damage to the body diode!

[image]

As shown on the right, the body diode cathode connects to the NMOS drain.

kb1lqc, 2017-02-15:
**Faraday Rev B SN 3**
This is using an old firmware but should be mostly relevant. This radio has an Antennova GPS installed.

_Nominal VCC Connection @ 9V_
Steady State Current (Boot-up): 22mA
Steady State Current (GPS Lock): 21mA

_MOSFET VCC Connection @ 9V_
Steady State Current (Boot-up): 23mA
Steady State Current (GPS Lock): 24mA

Yup! As expected Faraday turned on when VCC cable was plugged into the MOSFET connection! Hell yeah @kb1lqd! This is only for correctly oriented voltages. However in aerospace terms we're already at one fault :) and therefore are one fault tolerant hehe.

kb1lqd, 2017-02-15:
Connect to USB at same time? I suspect nothing interesting...

kb1lqd, 2017-02-15:
Also, make sure to confirm on a REV D1.

kb1lqc, 2017-02-15:
While 9V is connected to the MOSFET connector, I measured 8.33V or 670mV less than the power supply. Exactly where I'd expect it to be since that's a silicon diode drop!

kb1lqc, 2017-02-15:
Awesome, confirmed that while VCC connected to MOSFET connector the USB connection still works. At least it identifies as a serial port in Windows.

[image]

kb1lqc, 2017-02-15:
Removing VCC from MOSFET while USB connected showed no signs of alternate operation other than intended smooth power switching.

kb1lqc, 2017-02-15:
I put 17V on the power supply and connected in to both power and MOSFET connectors, unit operated as expected in these conditions. I did hear an audible click with my Rev B unit but likely the TPS562201DDCR ramping up.

kb1lqc, 2017-02-15:
Placing 17V reverse polarity across VCC input resulted in no damage. Faraday was obviousy non-functional but restoration of USB power or VCC power with correct polarity after the fault showed nominal operation.

I would however caution that I did not test inverted VCC power with USB connected. While this should be OK with a floating VCC (non-grounded battery or isolated power supply) there could be problems if the external supply is referenced to the same ground as the computer. If the computer is floating (laptop on battery power) then you should be good to. However these are not tested. I simply don't want to try that right now on my computer.

kb1lqc, 2017-02-15:
**Rev D1**

I placed 17V across VCC and all worked fine. Additionally, 17V across the MOSFET connector also worked fine. Both LEDs blinked during bootup in both cases. I did not measure current and this unit did not have a GPS (SN 40) but that isn't relevant to this test. We shouldn't expect to see immediate failures from an incorrect connection.

kb1lqc, 2017-02-15:
Alright I'm done. I tested Rev D1 SN 40 with reverse polarity 17V (isolated) into both VCC and MOSFET connections. This resulted in no observable power supply current (rough, no DMM) and when 9V was reapplied from a battery correctly oriented nominal operation resumed.

Users should be able to plug an isolated (bench-top) supply or 9V battery in backwards or into MOSFET and not cause damage in this single-fault scenario.

With two faults such as reverse polarity VCC and a power supply plugged into MOSFET we believe nothing should happened due to the LTC4412 but cannot guarantee it and it's untested. We can only protect you from so much. So far, I like the design, Faraday connectors for VCC and MOSFET are reversible without causing any obvious damage. Plug and play without having different connectors or oddly placed connectors.

closed 2017-02-15 by kb1lqc