FaradayRF/FaradayRF-Hardware #51: Generate 3-view Mechanical Drawing of PCB with Components

markwiem, 2017-07-09:
Ideally this would be viewable in the `FaradayRF-Hardware/Mechanical/Drawings` (or similar) folder without the need to pull down zips, enter KiCad, etc. Critical to enthusiasts who want to know how to mount FaradayRF in an enclosure. 

I noticed the hole patter of the FaradayRF is not regular (ie 4"x4", etc.). I would be happy to generate a 3D printed adapter (`FaradayRF-Hardware/Mechanical/CAD/Mounts`) to transverse the irregular PCB mounts to an "easy to measure out and manufacture at home" variety. 

Later revisions of the board may want to consider evolving mounts to a symmetrical pattern. I completely understand the challenge of doing this real-world.

kb1lqc, 2017-07-09:
Yes this should be a thing! Thanks for bringing this up. One note, the hole pattern is purposefully irregular so you can't get mounting orientation backwards. An adapter would definitely be useful too for uses when this isn't an issue.

kb1lqd, 2017-07-09:
I will second or third it that at least a mechanical drawing about to
output hole patterns would be great. I've eye baked it when drilling
mounting holes in my base station for Faraday but this is not easy or high
quality.

What tools do you think we can use?

markwiem, 2017-07-10:
I know that some PCB layout tools provide CAD export in `.STEP` or `.stl` format. Looking at KiCad, it doesn't seem that this can be performed natively, but there is a StepUp tool that could potentially work for us here (https://forum.kicad.info/t/how-to-share-step-3d-model-with-all-the-components-on-the-pcb-too/2542). 

If StepUp worked, I should be able to import the model into a CAD suite and generate basic drawings with dimensions. I'm not going to go as far as to say having a PCB CAD export is required for this, as we could could generate decent enough dimensional drawings from high res pics of the board. 

In terms of generating a mounting adaptor, I'm proposing using Fusion 360. It is free for makers as long as income restrictions are met (see: https://www.autodesk.com/products/fusion-360/overview -- scroll to "Collaborate and Manage"). There would be no licensing fees associated with the board, if it ever looked like volume/revenue could make this a (good!) problem, then you could whip up a clone in a licensed version and be set. Let me know if you feel comfortable with this strategy.  

To get started on the mounting adaptor, a basic photo with measure overlayed would be sufficient, it could even be a napkin drawing dropped in this thread :)

kb1lqc, 2017-07-11:
Yeah Faraday Hardware is currently designed on the older Kicad 2013 Stable too... it's a huge deal for me to update the libraries. Figured next rev of the board.I wonder if it's possible to import at least the gerbers or something into a CAD tool. I know I don't have many 3D parts in the 2013 library.This makes me look forward to moving to the new Kicad. It will definitely be more stable than when we designed this board 😉---- On Sun, 09 Jul 2017 21:15:12 -0700  Mark Wiemer<notifications@github.com> wrote ----I know that some PCB layout tools provide CAD export in .STEP or .stl format. Looking at KiCad, it doesn't seem that this can be performed natively, but there is a StepUp tool that could potentially work for us here (https://forum.kicad.info/t/how-to-share-step-3d-model-with-all-the-components-on-the-pcb-too/2542). If StepUp worked, I should be able to import the model into a CAD suite and generate basic drawings with dimensions. I'm not going to go as far as to say having a PCB CAD export is required for this, as we could could generate decent enough dimensional drawings from high res pics of the board. In terms of generating a mounting adaptor, I'm proposing using Fusion 360. It is free for makers as long as income restrictions are met (see: https://www.autodesk.com/products/fusion-360/overview -- scroll to "Collaborate and Manage"). There would be no licensing fees associated with the board, if it ever looked like volume/revenue could make this a (good!) problem, then you could whip up a clone in a licensed version and be set. Let me know if you feel comfortable with this strategy. To get started on the mounting adaptor, a basic photo with measure overlayed would be sufficient, it could even be a napkin drawing dropped in this thread :)